TimSchneider42/franky #7: [Communication_constrains_violation] error

YeqingqingHuang, 2024-04-03:
I have been encountering an error randomly when attempting to move the robot: “Move command aborted: motion aborted by reflex! [“communication_constrains_violation]” This error does not occur every time, but it happens quite frequently. If I reactivate the robot and perform a Cartesian linear movement, the robot can run successfully without any error messages. However, if I try to repeat this movement without reactivate the robot, this error message will be triggered during the movement and the robot will be stopped. 

I have already run the simple ping test and communication test as mentioned in the document. It doesn’t seem to be a problem. 

The result of simple ping test:

```
10000 packets transmitted, 10000 received, 0% packet loss, time 10007ms
rtt min/avg/max/mdev = 0.111/0.211/0.478/0.042 ms
```
The result of communication test:

```
Control command success rate of 9978 samples:
Max: 1.00
Avg: 0.99
Min:0.94
```

I have also disabled the CPU frequency scaling, here is the output (The maximum frequency is 3.90 GHz):
`Current CPU frequency is 3.77 GHz.`

TimSchneider42, 2024-04-03:
Hi, 

this error usually appears when you are not running on a realtime kernel. Could you confirm that you are using a realtime kernel (`uname -a`)? Also, make sure that the executing user has permissions to execute realtime applications as described in the [libfranka documentation](https://frankaemika.github.io/docs/installation_linux.html#allow-a-user-to-set-real-time-permissions-for-its-processes).

Best,
Tim

TimSchneider42, 2024-04-03:
Hi again,

I just updated the README with instructions on how to install a real-time kernel.

Best,
Tim

YeqingqingHuang, 2024-04-05:
Hi, thank you for your answser. 

I have alread implemented what you suggested, but the error `Communication_constrains_violation` still occurs sometimes randomly. 

Additionally, there is another error during the movement:
`franky._franky.ControlException: libfranka: Move command aborted: motion aborted by reflex! ["cartesian_motion_generator_joint_velocity_discontinuity", "cartesian_motion_generator_joint_acceleration_discontinuity"]`

So these two errors occur quite often when I restart my program without changing any code and reactivating the robot

TimSchneider42, 2024-04-05:
Hi, 

can you confirm that the right kernel is loaded with `uname -a`?

Also make sure that you do not see a log message like this when running your program: ```libfranka: unable to set realtime scheduling: Operation not permitted```

Regarding you second error, this is unfortunately something coming from libfranka that happens from time to time, especially when the robot is in contact with something. Try reducing the dynamics a little bit  (see `robot.relative_dynamics_factor`).

Best,
Tim

YeqingqingHuang, 2024-04-05:
Hi Tim,

after typing `uname -a` in the terminal, I received the following output:
`Linux [PCNAME] 5.15.148-rt74 #2 SMP PREEMPT_RT Tue Mar 19 12:02:21 CET 2024 x86_64 x86_64 x86_64 GNU/Linux`
And I have not received such a log message when running my program. 

The `Communication_constrains_violation` error occurs almost every time, when I try to make the robot perform linear movements repeatedly, such as moving up and down multiple times.  Could this be a problem?  Because with a simple linear movement the robot can complete it successfully without any error messages.

TimSchneider42, 2024-04-05:
Hi,

your kernel looks good. How is your network setup? Could you try connecting the robot and the computer directly?

Best,
Tim

YeqingqingHuang, 2024-04-05:
Hi, 
the robot is already connected with the computer directly. Could the reason for this error lie in the unstable network?

TimSchneider42, 2024-04-05:
That's what I was thinking. But if the connection is direct, that is unlikely to be the issue. We observed this issue once when the CPU load on the control PC was really high. Maybe this is something you could check as well if the issue reappears.

Anyway, I am glad it works now!

Best,
Tim

YeqingqingHuang, 2024-04-16:
Hi Tim, 

the error `communication_constraints_violation` is still occurring frequently only when performing a movement with the help of `CartesianMotion()`. I still can't figure it out. I need to have a repeatable linear movement of the robot. To achieve this, I used a for-loop. Could this be a problem?

TimSchneider42, 2024-04-16:
Hi,

I don't think this issue is caused by python code. I feel like this should be something more low-level. Is your CPU load high?

Best,
Tim

YeqingqingHuang, 2024-04-16:
Hi, 

While implementing the motion, I observed that the CPU utilization reaches a maximum of 65%. However, this error could also come out with low CPU usage. 
I also tried the `JointWayPointMotion` function, in this case, the robot can work really well.

TimSchneider42, 2024-04-16:
To investigate this issue further, you could try registering motion callbacks:
```python
data = []
def callback(robot_state, time_step, rel_time, abs_time, control_signal):
    data.append((time.time(), rel_time, abs_time))

motion.register_callback(callback)

# Plot the collected data after the motion is finished
```
This function will be called whenever the controller is updated (1kHz). Maybe it is possible to see some delays in there. Also, what kind of CPU are you using?

Best,
Tim

YeqingqingHuang, 2024-04-16:
I am using "Intel(R) Core(TM) i3-8145U CPU @ 2.10GHz"

TimSchneider42, 2024-04-16:
Maybe your processor could be part of this issue. So far, I have only tested this code on two different i9 processors ("Intel 11th Gen Core i9-11900K" and "Intel Core i9-10900K"). Do you have access to a more powerful CPU for testing by any chance?

YeqingqingHuang, 2024-04-17:
I tried with the other CPU "Intel(R) Core(TM) i5-8250u CPU @ 1.60GHz". The error didn't occur every time as before, but it still happened randomly. 

Besides this 'communication' error, sometimes I could also get other errors, such as `cartesian_motion_generator_joint_acceleration_discontinuity.` or `cartesian_motion_generator_joint_velocity_discontinuity`, `cartesian_motion_generator_joint_acceleration_discontinuity`. I set the relative_dynamic_factor to 0.03. Sometimes the robot could work really well, but these errors could happen suddenly.

TimSchneider42, 2024-04-17:
Alright, then it is likely related to your CPU load. You could try playing around with the CPU affinity/cgroups to reserve a specific core purely for your process, but I am unfortunately also not an expert on that.

Regarding the discontinuity errors you are getting, this is, unfortunately, a known issue with the cartesian control mode of Franka robots. In my experience, it got better once I used a Franka Research 3 and upgraded its firmware, but it never fully disappeared. In my case, I implemented automatic error recovery and just lived with it.

TimSchneider42, 2024-04-17:
No worries, happy to help!

closed 2024-04-17 by TimSchneider42

mahaarbo, 2024-06-03:
In case this is relevant to others, we had the same issue with `communication_constraints_violation` popping up quite consistently with the PyPi release of franky (on i5-1240P and slower CPUs). What worked for us was to build libfranka with the Release target as this had been reported as a possible fix, and use the `-O3` optimization flag on franky (with Release of course), though we don't know if the optimization is strictly necessary. Does the PyPi release pipeline compile libfranka with the Release target?
This completely eliminated the communication constraints violation, and resulted in a relatively stable system (except for the known intermittent Cartesian motion generator issues).

TimSchneider42, 2024-06-03:
For the PyPi release, I set `CMAKE_BUILD_TYPE=Release` when compiling franky (which does enable -O3 by default). However, I just noticed that libfranka itself had not been built in release mode. I have fixed that now, and I think I will soon upload a new version to PyPi.

Thanks a lot for your comment!

TimSchneider42, 2024-06-04:
I just uploaded a new release to PyPi (0.9.0). In this version, libfranka is built with compiler optimizations. However, it also includes some backward-compatibility breaking changes in the gripper API. Let me know if that fixes your problems once you get a chance to test it.

mahaarbo, 2024-06-06:
We can confirm that the new update works, tested with Ubuntu 22.04 with preempt_rt, (i5-1240P cpu), and ran a couple of Cartesian motion commands repeatedly for at least 30 minutes. No communication constraint violations occurred.

moritz-surag, 2024-07-05:
would the PyPi version also work with the older Franka Emika Panda? If not, is the libfranka build with compiler optimizations also included in the wheels that you provide for Franky 0.9.1?

TimSchneider42, 2024-07-09:
For the Panda you require libfranka 9.2.0, which is not in PyPi. Just follow the instructions in the [README](https://github.com/TimSchneider42/franky/?tab=readme-ov-file#installing-franky) and install the wheels directly from the repo. In these wheels, compiler optimizations are turned on.

moritz-surag, 2024-07-27:
thanks for the feedback. I can confirm that upgrading to Franky 0.9.1 fixes the communication constraints violation error on Intel Nuc with i5-8365U CPU running Ubuntu 20.04.6 LTS with RT Kernel. Thanks!