TimSchneider42/franky #9: The robot do jerky sound.

chaitanyantr, 2024-06-14:
**Issue:** The franka emika robot doing jerky sounds.

**when** the robot is moving from one waypoint to other way point.
**Observations**:- 5 different robots in my lab do same sounds.(probably not a hardware issue)
**what code?:-**  The default code of moving from m1 to m2.
What language:- python sample code 

```
# A point-to-point motion in the joint space
m1 = JointWaypointMotion([JointWaypoint([-0.3, 0.1, 0.3, -1.4, 0.1, 1.8, 0.7])])

# A motion in joint space with multiple waypoints
m2 = JointWaypointMotion([
    JointWaypoint([-0.3, 0.1, 0.3, -1.4, 0.1, 1.8, 0.7]),
    JointWaypoint([0.0, 0.3, 0.3, -1.5, -0.2, 1.5, 0.8]),
    JointWaypoint([0.1, 0.4, 0.3, -1.4, -0.3, 1.7, 0.9])
])

robot.move(m1)
robot.move(m2)

```

TimSchneider42, 2024-06-18:
Hi, 

Which robot are you using?

I noticed this behavior, too, with our Franka Panda in joint position control mode. What seemed strange to me was that even when I commanded the current joint position `RobotState.q` as a target, these noises were audible. However, I did not really have time to look into this issue much further.

If you would like to investigate this issue further, I suggest doing two things:
1. Test whether the issue persists with older franky versions (like 0.8.0)
2. Profile the commanded and actual joint trajectories of the robot. You can achieve this by using `Motion.register_callback` as such before executing the motion:
```python
traj = []

def cb(robot_state: franky.RobotState, time_step: float, rel_time: float, abs_time: float, control_signal: franky.ControlSignal):
    # This function will get called for every control update the robot does (1kHz).
    # Time, current pos, desired pos, current vel, desired vel, desired acc, next commanded pos
    traj.append((abs_time, robot_state.q, robot_state.dq, robot_state.dq, robot_state.dq_d, robot_state.ddq_d, control_signal.q))

m1.register_callback(cb)
robot.move(m1)

# Plot the trajectories, e.g. with matplotlib
```

I am happy to help if you need any assistance with this.

Best,
Tim

weristeddy, 2025-01-15:
I am currently also facing this problem using franky with libfranka 0.8.0. Furthermore i observe that the robot is stopping between the waypoints even when the whole JointWaypointMotion object is passed into the move() method. Is the the expected motion when using waypoints? My goal is to drive a fluent trajectory.

TimSchneider42, 2025-01-17:
Hi, 

Stopping at the waypoints is expected. Unfortunately, the free version of the ruckig motion planning software does not support automatically calculating optimal velocities for intermediate waypoints. Hence, those have to be set manually and default to 0. What you can do is set the target velocity of the intermediate waypoints to a value different from 0, as shown in the [Readme](https://github.com/TimSchneider42/franky?tab=readme-ov-file#motion-types) (motion m3). You could calculate those waypoint velocities with a spline interpolation, e.g., using [scipy](https://docs.scipy.org/doc/scipy/reference/generated/scipy.interpolate.CubicSpline.html#scipy.interpolate.CubicSpline).

Regarding the jerky sound, I remember spending some time investigating it. Strangely, the robot does this even when it is commanded just to hold its current position, and it only does it in the first motion of a sequence of waypoints. I assume it is somehow related to the robot switching to the joint controller. 

Best,
Tim

ferrolho, 2025-02-09:
Hi, @TimSchneider42. In the code snippet above for registering the callback, you mentioned:
> This function will get called for every control update the robot does (1kHz).

I am trying to understand what is the control frequency used by the `JointWaypointMotion`. In other words, what is the `dt` between each waypoint? I have read the README.md fully, but I couldn't find a mention to this. Thanks!

TimSchneider42, 2025-02-09:
Hey,

The time between waypoints is not static. The Ruckig motion planner will always compute a time-optimal trajectory between waypoints within the configured jerk, acceleration, and velocity limits.

Best,
Tim

closed 2025-04-02 by TimSchneider42