TimSchneider42/franky #10: set_cartesian_impedance not affecting robot stiffness

purpleyoda1, 2024-06-28:
Hi!
I'm experiencing an issue with the `set_cartesian_impedance` function while working with an Emika Panda robot. The function doesn't seem to be changing the robot's stiffness at all, regardless of the input values. It remains as stiff as ever. The set_joint_impedance however works as expected. 

`def cartesian_imedance_test(robot: Robot):
    robot.set_cartesian_impedance([10., 10, 10, 200., 200, 200])
    input("press enter to reset impedance")
    robot.set_cartesian_impedance([2000., 2000, 2000, 200., 200, 200])

def joint_imedance_test(robot: Robot):
    robot.set_joint_impedance([10., 3000, 3000, 10, 3000, 3000, 3000])
    input("press enter to reset impedance")
    robot.set_joint_impedance([3000.]*7)

def main():
    IP = "192.168.3.10"
    robot = Robot(IP)
    gripper = Gripper(IP)
    robot.relative_dynamics_factor = 0.1
    robot.recover_from_errors()
    robot.set_ee([1.0, 0.0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, 0.0, 0.0, 1.0])

    m = to_home()
    robot.move(m)

    #joint_imedance_test(robot)
    cartesian_imedance_test(robot)

if __name__ == "__main__":
    main()`

Any thoughts about what is wrong? Or what i can do to try to identify/fix the issue?

TimSchneider42, 2024-06-28:
Hi,

I am not sure I understand your code correctly. Do you just want the robot to use impedance control to hold its current position? It might be that the impedance parameters only have an effect while the robot is in control (that is, executing a motion). 

What you could try is executing an asynchronous waypoint motion, where you set the `minimum_time` parameter of the waypoint to ensure that the motion keeps going (though I have never tested this parameter). 

Another option could be the `ExponentialImpedanceMotion` class, which I have also never tested.

Please let me know how it goes.

Best,
Tim

purpleyoda1, 2024-07-02:
Thank you! Yes i just wanted it to hold the position while i physically checked for changes in the impedance by trying to move the robot. Will test the functions you suggested and come back to you

closed 2024-12-02 by TimSchneider42