TimSchneider42/franky #11: Guarded motions aborted by joint velocity/acceleration discontinuity

purpleyoda1, 2024-06-28:
Hi again!

**What:** Motions being stopped by joint velocity or joint acceleration discontinuity

**When:** While using guarded motions to probe a surface for environment calibration.

**In detail:** Motions are being aborted when there seems to be no reason for it, by a Reflex triggered by either joint velocity, joint acceleration, or both. This seems to happen either when starting or stopping the motion. Im using a relative_dynamics_factor of 0.1 for the overall motions, and again for the guarded probing motion, meaning it's all the way down to 0.01. The speeds really shouldnt trip the robot up, and this is corroborated by printing the joint velocities while running, as they rarely exceed 0.1. It should not be the forces acting on the robot either, as printing them while running show that they are of reasonable values and within (or over when it hits something) given limits of the movements. 

Questions: Can this be caused by scaling the speed down so much? Does the limits also get scaled down, and does the cartesian_motion_planner takes these adjusted speeds and limits into account?
Is there something that can be done by altering the collision_behaviour? 
Any other thoughts about what to look for or change?

Any and all help is greatly appreciated! Thanks!

TimSchneider42, 2024-06-28:
Hi,

Setting relative_dynamics_factor does not influence the collision thresholds. It usually helps to increase the thresholds with `robot.set_collision_behavior`; however, in my experience, these errors never fully disappear.

Best,
Tim

purpleyoda1, 2024-07-02:
I see, i will try experimenting with the set_collision_behavior then. Thanks!

closed 2024-07-02 by purpleyoda1