TimSchneider42/franky #12: Absolute cartesian position control keeps falling

Topasm, 2024-07-15:
`
def move_robot(dx=0.0, dy=0.0, dz=0.0, drot=None):
    current_pose = robot.current_cartesian_state.pose
    end_effector_pose = current_pose.end_effector_pose  # Access the end_effector_pose

    # print("Current end effector pose:")
    # print("Translation:", end_effector_pose.translation)
    # # Extract translation and rotation from end_effector_pose
    translation = np.array(end_effector_pose.translation)
    rotation = np.array(end_effector_pose.quaternion)

    new_translation = translation + np.array([dx, dy, dz])

    if drot is not None:
        current_rotation = R.from_quat(rotation)
        delta_rotation = R.from_euler('xyz', drot)
        new_rotation = (delta_rotation * current_rotation).as_quat()
    else:
        new_rotation = rotation

    # Correctly format the translation and quaternion for Affine constructor
    motion = franky.CartesianMotion(
        franky.Affine(new_translation, new_rotation), franky.ReferenceType.Absolute)
    robot.move(motion, asynchronous=True)
`

I am running this function in while loop. This code get current postion and calculate new position with current postion + diff.
What I want do is move end-effector in  absolute position. 

But in test it's end-effector doen't keep its postion. I change end effector weight in desk but with small amount of disturb(weight) it doesn't keep its postion.

can I change stiffness of position in cartesian control?
I know it could solve with relative control mode but I need this way too.

TimSchneider42, 2024-07-15:
Hi,

I am not sure I understand your issue. Are you commanding a delta position that the robot does not reach?

Best,
Tim

Topasm, 2024-07-15:
No,
If simply
Loop{
get current_ee_position
Move(current_ee_position)
}

What i expected is keeping its position but EE keeps falling

Thanks for feedback Tim

TimSchneider42, 2024-07-15:
My guess would be that the end-effectors weight is configured lower than it actually is (in the web interface). Also by getting the current position in every step, you will have compounding errors which certainly cause the robot to slowly diverge from its original position. If you really want to hold the current position, I recommend just getting the position once in the beginning. Also you might check out my suggestion [here](https://github.com/TimSchneider42/franky/issues/10#issuecomment-2196456057).

Topasm, 2024-07-16:
I tried
``` python
motion = franky.ExponentialImpedanceMotion(
            affine, franky.ReferenceType.Absolute, 200, 100)
```
but it makes
```
franky._franky.ControlException: libfranka: Move command aborted: motion aborted by reflex! ["communication_constraints_violation"]
control_command_success_rate: 0 packets lost in a row in the last sample: 9
```

this error.
 Anyway, with just getting the position in the beginning I could make robot hold  z position well.

thanks Tim

closed 2024-07-16 by Topasm