TimSchneider42/franky #15: Problem with gripper movement

OXOSMDF, 2024-10-16:
Hello, I try to use franky to manipulate franka panda and collect data from it. I wrote this ROS node. I separate gripper control, robot movement control and data collection into 3 threads. I found problem when I use gripper.open_async and gripper.grasp_async as I would block other 2 threads and I could only use other 2 threads after open gripper and grasp gripper is done. 
`#!/usr/bin/env python
import rospy
from geometry_msgs.msg import Twist
from franky import *
import franky
import threading
from std_msgs.msg import Int16, Float32
from teng_arduino.msg import TwistGripper
import pyrealsense2 as rs
import numpy as np
import pickle
import copy
import traceback
import os
from scipy.spatial.transform import Rotation
import open3d as o3d

def visualize_point_cloud(points):
    """
    Visualize the point cloud using Open3D.
    :param points: The Nx3 numpy array of point cloud coordinates.
    """
    # Convert points to Open3D point cloud
    pcd = o3d.geometry.PointCloud()
    pcd.points = o3d.utility.Vector3dVector(points)

    # Visualize the point cloud
    o3d.visualization.draw_geometries([pcd], window_name="Point Cloud Viewer",
                                      width=640, height=480)
    
def quaternion_to_euler(q):
    
    q_x, q_y, q_z, q_w = q[0], q[1], q[2], q[3]
    roll = np.arctan2(2.0 * (q_w * q_x + q_y * q_z), 1.0 - 2.0 * (q_x**2 + q_y**2))
    pitch = np.arcsin(2.0 * (q_w * q_y - q_z * q_x))
    yaw = np.arctan2(2.0 * (q_w * q_z + q_x * q_y), 1.0 - 2.0 * (q_y**2 + q_z**2))

    return roll, pitch, yaw

class FrankaRobotController:
    def __init__(self):
        rospy.init_node('franka_robot_controller')
        robot_ip = rospy.get_param('~robot_ip')
        self.robot = Robot(robot_ip)
        self.robot.recover_from_errors()
        self.robot.relative_dynamics_factor = 0.05
        self.robot.velocity_rel = 0.4
        self.robot.acceleration_rel = 0.2
        self.robot.jerk_rel = 0.02
        self.gripper = franky.Gripper(robot_ip)
        self.gripper_speed = 0.02
        self.gripper_force = 5
        print('Connecting to robot at {}'.format(robot_ip))
        
        target_positions = [0.2, 0, -0.28, -2.63, -0.03, 2.61, 0.75]
        joint_state_target = JointState(position=target_positions)
        waypoint = JointWaypoint(target=joint_state_target, reference_type=ReferenceType.Absolute)
        init_motion = JointWaypointMotion([waypoint])
        self.robot.move(init_motion)
        
        # self.twist_sub = rospy.Subscriber('/franka_twist', Twist, self.twist_callback)
        self.gripper_sub = rospy.Subscriber('/franka_gripper_command', Int16, self.gripper_open_callback, queue_size=1)
        self.twist_gripper_sub = rospy.Subscriber('/franka_twist_gripper', TwistGripper, self.twist_gripper_callback, queue_size=1)        
        self.translation = [0, 0, 0]
        self.angular_y = 0.0
        self.movement = [0, 0, 0, 0]
        self.new_command_received = False
        self.new_grippr_command_received = False
        self.gripper_cnt = 0
[... 117 lines cut ...]
                
        rate.sleep()
    
    def log_current_thread(self):
        current_thread = threading.current_thread().name
        rospy.loginfo(f'Running thread: {current_thread}')

    def run(self):
        try:
            #translation = np.array([[0.0], [0.1], [0.0]])
            #quaternion = np.array([[0.0], [0.0], [0.0], [1.0]])
            #way = Affine(translation=translation, quaternion=quaternion)
            #motion_forward = CartesianMotion(way, ReferenceType.Relative)
            #self.robot.move(motion_forward, asynchronous=True)
            #motion_backward = CartesianMotion(way.inverse, ReferenceType.Relative)
            #self.robot.move(motion_backward, asynchronous=True)
            self.gripper.move_async(self.gripper.max_width/2, self.gripper_speed)
            print('Gripper opened: {}'.format(self.gripper.max_width))
            rospy.spin()
        finally:
            self.save_data()

if __name__ == '__main__':
    controller = FrankaRobotController()
    controller.run()`

Xiaoyao-Li, 2025-02-26:
Hi,
Thanks for the awesome franky, which allows me to easily control my Franka🚀
I also found that asynchronous control of the gripper and the arm **within the same process** can cause conflicts, even when using multithreading. 
However, I discovered that this issue can be resolved when the gripper and the arm **are controled in separate processes**. When I run two Python scripts simultaneously—one for controlling the arm and the other for the gripper—the low-frequency Franka gripper movements no longer block the high-frequency Franka arm control.

Best,
Puhao

yaoxt3, 2025-02-26:
Hi,

frankly thanks for your project. In the end, I figure out some way to control the robot arm and gripper at the same time in multithreading.
 `def gripper_control_loop(self):
        rate = rospy.Rate(2)
        while not rospy.is_shutdown():

            if self.new_gripper_command_received:
                if current_gripper_move:
                    current_future = self.gripper.open_async(self.gripper_speed)
                elif not current_gripper_move:
                    current_future = self.gripper.grasp_async(0.03, self.gripper_speed, self.gripper_force, epsilon_outer=0.2)
                    
            rate.sleep()`
I don't no why but it actually works. Thanks again for your contribution.

Best regards,
Yirui

TimSchneider42, 2025-02-28:
@Xiaoyao-Li, I implemented a solution for the future issue in version 0.11.2. Could you give it a go and let me know if it works?

TimSchneider42, 2025-03-02:
Thanks for testing and letting me know @Xiaoyao-Li! Also, I appreciate the offer. Might come back to it later on :)

closed 2025-03-02 by TimSchneider42