TimSchneider42/franky #16: Motion Command Aborted

sesem738, 2024-10-31:
Hi,

I am trying to use the franky library to execute my trained RL policies. However, I cannot run any of the franky examples, though I can run the examples from the libfranka library. I am using a laptop with a 7th Gen Core i7 and have set the power mode to performance. When I run the communications test, it has a 100% success rate. I am using version 0.9.1 of franky. Does anyone have advice on how to rectify this issue?

[image]

TimSchneider42, 2024-10-31:
Hi,

as you can see in the error message, you have a joint limit violation. Try to move the robot manually into a pose in which none of the joints are at their limit and make sure that you do not command the robot to move into the joint limits.

Best,
Tim

sesem738, 2024-10-31:
Hi, 

Thanks for the response. I have tried different initial positions before I run the code, but I am still getting the same error. I have add a picture of the initial position I use. I am also wanted to add that I am using the FRANKA RESEARCH 3. 
[image]

TimSchneider42, 2024-10-31:
Can you check whether the second joint from the bottom is close to its joint limit?

closed 2024-12-02 by TimSchneider42