TimSchneider42/franky #17: Change to guiding mode

weristeddy, 2024-11-01:
Hello, I am wondering why i am not able to change the RobotMode?
I tried using the set_guiding_mode() function from the Robot class but my RobotMode after checking was still in Idle. Same also for the stop() function.

weristeddy, 2024-11-01:
maybe good to know: i am using the Franka Emika Robot with libfranka 0.8.0

TimSchneider42, 2024-11-01:
Did you press and hold the two buttons on the robots gripper?

weristeddy, 2024-11-01:
yes, then a kind of purple light appears. Guiding mode some how only works if i press the black stop button before and then press the buttons for driving mode

TimSchneider42, 2024-11-01:
Maybe you have to press the button to move it? I think the `set_guiding_mode` function only sets which dimensions can be guided in guiding mode.

weristeddy, 2024-11-02:
okay, so there is no way to change the RobotMode from 'Idle' to either 'UserStopped' or 'Guiding' using only code? 
Are these modes only accessible when pressing manually the stop and gripper buttons?

TimSchneider42, 2024-11-02:
To be honest, I am not a 100% sure how it works. I remember playing around with the function once and being able to lock rotation but not translation. I think I still had to press the stop button. Maybe it is best to ask in the Franka forum.

weristeddy, 2024-11-04:
alright, thank you for your help :)
Do you maybe know what the robot.stop() function does?

TimSchneider42, 2024-12-02:
Sorry for the late reply, I was on vacation. The stop function terminates the current motion and changes the robot's state back to idle. The userstopped mode is when the black button is pushed, I believe.

closed 2025-02-26 by TimSchneider42