TimSchneider42/franky #18: Segmentation fault (core dumped)

HaoyiZhu, 2024-12-02:
Hi, thanks for the amazing project!

After I build franky, seems that it always gives out `Segmentation fault (core dumped)` error. For example:

```bash
$ python -c "from franky import Affine; Affine([0.2, 0.0, 0.0])"
[1]    1416578 segmentation fault (core dumped)  python -c "from franky import Affine; Affine([0.2, 0.0, 0.0])"
```

Could you help me to find the problem? Thanks!

TimSchneider42, 2024-12-02:
Did you make sure that the correct library versions are installed? Also, note that we provide pre-built wheels. So as long as you just want to use the python interface, use pip to install franky.

TimSchneider42, 2024-12-02:
Maybe to add to that, I wrote a docker container for the build process. Maybe you can just use that. See the docker-compose file in the project root.

HaoyiZhu, 2024-12-02:
Hi, thanks for your quick response!

I am using Ubuntu20, Python 3.11.10, libfranka 0.13.3, Eigen 3.4.0 and Pybind11 2.11.1. Also, I've tried the pre-built wheels but still fail.

I will try docker later.

TimSchneider42, 2024-12-02:
If the pre-built wheels fail, then docker won't help either I assume. Does it already fail at the import or later when the Affine object is created?

HaoyiZhu, 2024-12-02:
It fails when the Affine object is created

TimSchneider42, 2024-12-02:
I can reproduce your issue. I assume you have installed numpy version 2, right? I think the issue is that the pybind version we are using does not support numpy 2. You can work around this problem by installing numpy 1.26.4. I will check for a better fix.

TimSchneider42, 2024-12-02:
Would you mind trying to build against pybind 2.12.0 or later?

HaoyiZhu, 2024-12-02:
Using numpy 1.26.4 solves my problem. However, pybind 2.12.0 + numpy 2.1.3 does not work.

Anyway, my problem is perfectly solved. Thank you so much for your swift response and fix!

TimSchneider42, 2024-12-02:
Great, no worries! Thanks for reporting the issue. I will try to fix this problem and publish a new version.

TimSchneider42, 2024-12-02:
I think I fixed the problem. The new wheels are currently building and should be available within the next 15min.

HaoyiZhu, 2024-12-02:
Great!

commit 230c6726 (2024-12-02): Updated pybind11 dependency version to 2.13.6 to support numpy 2. Fixes #18.
commit 247fe160 (2024-12-02): Updated pybind11 dependency version to 2.13.6 to support numpy 2. Fixes #18.