TimSchneider42/franky #19: Support for libfranka 0.14.x?

audreylee25128, 2025-01-14:
Hello,

How can I use franky with libfranka versions >0.14.0? The Franka Research 3's system version update to 5.7.1 required an upgrade from libfranka 0.13.3 to 0.14.0. I don't believe this libfranka version is covered by one of wheels from your latest release.

Thank you so much!

TimSchneider42, 2025-01-14:
Hi,

unfortunately, supporting libfranka 0.14 is not as straightforward as setting a new version number in the build script, as libfranka 0.14 needs pinocchio as an additional dependency. Hence, the build system has to be updated first. I am a bit busy, so I am unsure when I will find time to look into this. However, I would happily accept PRs if you or somebody else updated the build containers by adding the Pinocchio installation commands.

Best,
Tim

Zack-Ly, 2025-02-17:
My robot server version is 8, for libfranka, I must use 0.14.1 or higher. On this basis, I compiled the python wheel against version 0.14.1 via the franky source code, and the compilation process did not report any errors. After testing, it currently reads the robot state and controls the gripper motion properly, but the motion is not generated properly. For example, running motion = CartesianMotion(Affine([0.1, 0.0, 0.0]), ReferenceType.Relative) results in: segment error (core dumped). I'm wondering if it's because of the difference between libfranka 0.13.3 and 0.14.1 regarding Pinocchio that the source code needs corresponding changes at the same time?

Best wishes!

TimSchneider42, 2025-02-17:
@Zack-Ly, which pinnocchio version did you use?

Zack-Ly, 2025-02-17:
@TimSchneider42 Hello, first of all, thank you for your prompt reply！I have executed the pinocchio tutorial provided in the official franka library libfranka for installing version 0.14.0 and above, the tutorial is at https://stack-of-tasks.github.io/pinocchio/download.html, and installed pinocchio and its dependencies by “ sudo apt install -qqy robotpkg-py3*-pinocchio” . My pinocchio version as 3.3.0 under “/opt/openrobots/lib/pkgconfig”. My libfranka is currently at 0.14.1 on my system, and it works fine.   
Thank you for your amazing work, and I'm looking forward to reproducing your work in versions 0.14.0 and above!

TimSchneider42, 2025-02-17:
I am glad you like it!

That segmentation fault is strange. Did you use the Docker container for building or your local system? If you did not use the Docker container maybe try that. There is a guide for that in the Readme.

Unfortunately, I still don't have time to look into the issue myself. Maybe in two weeks from now...

Best,
Tim

TimSchneider42, 2025-02-25:
Hi @Zack-Ly and @audreylee25128,

I have just released version 0.11.0 of franky to PyPi, which adds support for libfranka 0.15.0. If you need 0.14.2, follow the instructions in the readme and download the pre-built wheels from GitHub.

Could you take a look at it and confirm that everything is working?

Best,
Tim

audreylee25128, 2025-02-25:
Thank you so much for taking the time to look into this! I'll give it a try in the coming days and keep you updated.

TimSchneider42, 2025-02-26:
Feel free to reopen this if you find any issues with the 0.14/0.15 support.

closed 2025-02-26 by TimSchneider42