TimSchneider42/franky #20: Question about FCI version.

EntrigesFichtenholz, 2025-03-06:
Hi,
First of all, thank you for this fantastic project! I have a question regarding the FCI version: Is a specific version required? I'm currently facing an issue where I cannot upgrade my Franka controller beyond version 2.1.4 because it is not cloud enabled. I haven't found a way to enable this functionality myself, and it appears that only Franka Emika can do so.

It would be wonderful if the project could work with older versions as well. I appreciate your time and help in advance!

TimSchneider42, 2025-03-06:
Hi,

I'm glad that you like the project! 

Regarding your question, you can try building the wheels for libfranka version 0.5.0, but I have never tried that, so there's no guarantee that it works. For that, you will need to install docker and docker-compose, and do
```bash
git clone https://github.com/TimSchneider42/franky/
cd franky
docker compose build franky-build --build-arg LIBFRANKA_VERSION=0.5.0
docker compose run --rm franky-build build-wheels
```

If that does not work, that likely means that franky uses some functionality that had not been implemented in libfranka 0.5.0. In this case, the best option is probably to find a way to upgrade your Franka to 4.2.1, which is supported by franky and install franky via
```bash
VERSION=0-9-2
wget https://github.com/TimSchneider42/franky/releases/latest/download/libfranka_${VERSION}_wheels.zip
unzip libfranka_${VERSION}_wheels.zip
pip install numpy
pip install --no-index --find-links=./dist franky-panda
```

Let me know if you need any further help!

Best,
Tim

EntrigesFichtenholz, 2025-03-07:
Wow, thanks for the fast response!
For others (and for me if I have to build it again): There is an error that indicates that CMake is trying to add a subdirectory for a dependency named ruckig (at /code/ruckig), but it can’t find a valid CMakeLists.txt there. This is typically because the ruckig dependency isn’t present or wasn’t properly initialized. If you got that, you run git submodule update --init --recursive
and enjoy the build process.

After that, follow the manual: $docker compose run franky-run bash
For now, I can get in the containers CLI and it seems to work. :)
I will test it tomorrow with the Panda.

TimSchneider42, 2025-03-07:
Good point, I forgot to add `--recurse-submodules`. I have edited the original answer.

EntrigesFichtenholz, 2025-03-17:
Only for record, you have to solve a merge conflict if you want to get it to run. Funny enough, it will compile with the newest version of libfranka to trick you of thinking that it was successful. Maybe I will get it solved someday... :)
For now Im stuck with the old Controllerversion 2.1.4 since you cant download a newer version, that wont work for cloud enabled controllers. 
Thank you for your support.

TimSchneider42, 2025-03-17:
I think here you should be able to download up to version 4.0.4 without using their cloud: https://support.franka.de/

EntrigesFichtenholz, 2025-03-17:
I have already tried it, unfortunately without success. :/

EntrigesFichtenholz, 2025-03-17:
Yeah, it will say that something is wrong with a merge conflict and will compile the newest version. Instead of 0.5.0. I will test it further when im back home.

EntrigesFichtenholz, 2025-03-17:
It seems that the merge conflict with libfranka 0.5.0 arises, because a file (control_tools.h) is not available in this version, while newer commits are based on it. At the same time, there are changes in the code (src/control_types.cpp) from different branches that cannot be merged automatically.

TimSchneider42, 2025-03-18:
I'm guessing the problem stems from the following line in the build script (`docker/common/install-libfranka`):

```bash
if ! git merge-base --is-ancestor f1f46fb HEAD; then git cherry-pick -m 1 f1f46fb; fi
```

This line cherry-picks a commit that adds some forgotten imports to `control_types.cpp` and `control_tools.h`. For some reason, this problem only occurred with later gcc versions. Hence, the Franka team did not notice it for a couple of versions. You could try commenting out that line and see if it works.

Regarding the upgrade to version 4.0.4, what do you mean by "decryption error"? We have upgraded a couple of our Pandas to version 4.0.4 that way, and everything worked fine. Of course, I cannot give you any guarantee that it will not cause you any problems.

Best,
Tim

EntrigesFichtenholz, 2025-03-18:
Unfortunately, it is a hardware error. We have already written to franka support. They also say that it can't work because the first generations of controllers don't support it. And we have one of those.  :(

Even with the 0.5.0 Libfranka build instruction, it seems that I compile with any newer version. Maybe my cusom Dockerfile dosnt work as it shoud.
Is there another way to check in the Buildet Docker container what version were running ? 

(env) root@IMEC009RobotPC:/home# python3 test.py 
Traceback (most recent call last):
  File "/home/test.py", line 3, in <module>
    robot = Robot("192.168.0.103")
franky._franky.IncompatibleVersionException: libfranka: Incompatible library version (server version: 3, library version: 9). Please check https://frankaemika.github.io for system updates or choose a libfranka version that uses the server version 3 from the table at https://frankaemika.github.io/docs/compatibility.html .

Here is my custom Dockerfile, I tried to beak in extra logic for the 0.5.0:

FROM quay.io/pypa/manylinux_2_34_x86_64:latest
LABEL maintainer="Tim Schneider <tim@robot-learning.de>"
RUN yum -y install poco-devel* zip boost1.78-devel urdfdom-devel fmt-devel
RUN ln -s /usr/lib64/cmake/Boost-1.78.0/ /usr/lib64/cmake/Boost

# Add the common directory but install dependencies manually
ADD docker/common/ /tmp/common/
ARG LIBFRANKA_VERSION

# Install Eigen3 (needed for libfranka)
RUN echo "Installing Eigen3..." && \
    git clone --depth 1 --branch 3.4.0 https://gitlab.com/libeigen/eigen.git /tmp/eigen && \
    cd /tmp/eigen && \
    mkdir build && \
    cd build && \
    cmake .. && \
    make install && \
    echo "Eigen3 installed successfully"

# Install pybind11 manually (extracted from install-dependencies)
RUN echo "Installing pybind11 version 2.12.0..." && \
    git clone --depth 1 --branch v2.12.0 https://github.com/pybind/pybind11.git /tmp/pybind11 && \
    cd /tmp/pybind11 && \
    mkdir build && \
    cd build && \
    cmake .. && \
    make install && \
    echo "pybind11 installed successfully"

# Install Catch2 manually (extracted from install-dependencies)
RUN echo "Installing Catch2 version 2.13.8..." && \
    git clone --depth 1 --branch v2.13.8 https://github.com/catchorg/Catch2.git /tmp/catch2 && \
    cd /tmp/catch2 && \
    mkdir build && \
    cd build && \
    cmake .. && \
    cmake --build . --target install && \
    echo "Catch2 installed successfully"

ARG PYBIN_SUPPORTED_VERSIONS="cp3(7m|8|9|10|11|12|13)"
RUN /tmp/common/manylinux-setup-pip

# Set working directory
RUN mkdir -p /code
RUN mkdir -p /build
VOLUME /code
VOLUME /build
WORKDIR /build
ADD docker/common/manylinux-build-wheels /usr/bin/
ADD docker/common/dependency_versions /usr/bin/
ADD docker/build/build-wheels /usr/bin/
ADD docker/build/run-tests /usr/bin/

# Clone libfranka, apply patch, and build
RUN git clone --recursive https://github.com/frankaemika/libfranka.git /tmp/libfranka \
    && cd /tmp/libfranka \
    && git fetch --all \
    && if [ "$LIBFRANKA_VERSION" = "0.5.0" ]; then \
         echo "Enforcing commit 5ca631e for libfranka 0.5.0"; \
         git reset --hard 5ca631e; \
         git clean -fdx; \
         git submodule foreach --recursive 'git reset --hard'; \
         git submodule foreach --recursive 'git clean -fdx'; \
       else \
         git checkout $LIBFRANKA_VERSION; \
       fi \
    && git submodule update --init --recursive \
    # Apply patch to fix missing include
    && sed -i '1i#include <stdexcept>' src/control_types.cpp \
    && mkdir build && cd build \
    && cmake -DBUILD_EXAMPLES=ON -DBUILD_TESTS=OFF -DCMAKE_CXX_FLAGS="-Wno-deprecated-copy" .. \
    && make -j"$(nproc)" \
    && make install

# Cleaning up
RUN rm -rf /tmp/common
RUN rm -rf /tmp/pybind11
RUN rm -rf /tmp/catch2
RUN rm -rf /tmp/eigen

# Preserve arguments for runtime
ENV PYBIN_SUPPORTED_VERSIONS=${PYBIN_SUPPORTED_VERSIONS}
ENV LIBFRANKA_VERSION=${LIBFRANKA_VERSION}

TimSchneider42, 2025-03-18:
Are you calling Docker build with `--build-arg LIBFRANKA_VERSION=0.5.0`? Also, why are you doing all the hard resetting? Does

```
RUN git clone --recursive https://github.com/frankaemika/libfranka.git /tmp/libfranka \
    && cd /tmp/libfranka \
    && git fetch --all \
    && git checkout $LIBFRANKA_VERSION \
    && git submodule update --init --recursive \
    && sed -i '1i#include <stdexcept>' src/control_types.cpp \
    && mkdir build && cd build \
    && cmake -DBUILD_EXAMPLES=ON -DBUILD_TESTS=OFF -DCMAKE_CXX_FLAGS="-Wno-deprecated-copy" .. \
    && make -j"$(nproc)" \
    && make install
```

not work?

In any case, add `RUN cd /tmp/libfranka && git status` to check if you are actually on the correct commit.

Best,
Tim

EntrigesFichtenholz, 2025-03-18:
I tried the hard resetting it because of the merge issue, also because of the std:: error which occurs after using with your suggestion.
For your second question: Yes, Im using those buildargs. But every time I run the final build and try to connect to my robot, I get the library version 9 feedback. Dont know why tho. 

I use exactly those build commands:
docker compose build franky-build --build-arg LIBFRANKA_VERSION=0.5.0
docker compose run --rm franky-build build-wheels

Even with my own dockerfile it says version 9.

The funny thing is, that my logs state that 0.5.0 is installed.

-- Installing: /usr/local/lib64/libfranka.so.0.5.0
-- Installing: /usr/local/lib64/libfranka.so.0.5
-- Installing: /usr/local/lib64/libfranka.so

This is, when I try my custom dockerfile.

With your other suggestion from previous comment:
When Im removing this line `if ! git merge-base --is-ancestor f1f46fb HEAD; then git cherry-pick -m 1 f1f46fb; fi`

It will crash on another stage.
The output shows that although CMake initially issues warnings during the build process (due to an outdated minimum version), the compilation process ultimately fails in several places. Specifically, in the control_types.cpp file, several functions attempt to use std::invalid_argument - for example to throw exceptions - without the corresponding <stdexcept> header being included.
The compiler points out that std::invalid_argument is defined in <stdexcept> and recommends adding this header.

The decryption error appears in the franka pandas controller interface. Therefore, the installation of any version that requires cloudenabled controllers wont install. therefore no 4.0.4 for me. :/

Thank you for your patience,
Johannes

TimSchneider42, 2025-03-19:
Hi Johannes,

This looks strange. It seems to me like this problem is coming from somewhere inside libfranka. Can you check if the same also happens when you request `robot.state`?

Best,
Tim

TimSchneider42, 2025-03-19:
To me, this feels like a bug in libfranka. In principle, you don't need the convenience functions I wrote (i.e., `robot.current_cartesian_state`, or `robot._current_pose`, ...), as you find all of this information in `robot.state` as well. Would that work for you?

EntrigesFichtenholz, 2025-03-19:
Yeah, I’m fine if its that what it takes. :)
For if its for example:
print(robot.state.q)
print(robot.state.dq)       
print(robot.state.cartesian_collision())

That woud be no problem. Right now I wrote a little test script, and it seems that I can readout the status of the arm, (Velocity Position and so on). But, robot.move(motion) raises also an error.

```
(rootenv) root@Fedora42:/home/johannes/Downloads/FrankyTest# python3 test.py 
Traceback (most recent call last):
  File "/home/johannes/Downloads/FrankyTest/test.py", line 18, in <module>
    robot.move(motion)
    ~~~~~~~~~~^^^^^^^^
ValueError: Invalid elbow configuration given! Only +1 or -1 are allowed for the sign of the 4th joint.

```
The Error comes from control_tools.h

```
/**
 * Checks if all elements of the elbow vector are finite and if the elbow configuration is valid
 *
 * @param elbow the elbow vector to check
 */
inline void checkElbow(const std::array<double, 2>& elbow) {
  checkFinite(elbow);
  if (!isValidElbow(elbow)) {
    throw std::invalid_argument(
        "Invalid elbow configuration given! Only +1 or -1 are allowed for the sign of the 4th "
        "joint.");
  }
}
```

TimSchneider42, 2025-03-19:
We will have to debug this properly to figure out what is going on. Can you maybe try some (print) debugging in libfranka to figure out where exactly the checkElbow function is called and with which value? Maybe this will help us figure out what is going on. 

Unfortunately, this problem does not appear in any of the newer libfranka versions I have tested so far, so I won't really be able to help you here. One thing that I would recommend, though, is to start with `robot.current_cartesian_state`, as it is a much simpler function than `robot.move`.

Let me know if you need any help.

Best,
Tim

EntrigesFichtenholz, 2025-03-20:
I looked at the libfranka versions 0.5.0 and 0.7.0. There were significant changes, but over all the elbow check should be the same. 
Also I ran a little python test script wich gave me this output:
```
envjohannes@Fedora42:~/Downloads/FrankyTest$ /home/johannes/Downloads/FrankyTest/env/bin/python /home/johannes/Downloads/FrankyTest/test.py
2025-03-20 12:25:14,299 - DEBUG - Initializing robot with IP: 192.168.0.103
2025-03-20 12:25:14,379 - DEBUG - Set relative_dynamics_factor: <franky._franky.RelativeDynamicsFactor object at 0x7f8ac90882f0>
2025-03-20 12:25:14,380 - DEBUG - Target translation: [0.2, 0.0, 0.0]
2025-03-20 12:25:14,383 - DEBUG - Created target Affine: Affine(t=[0.2 0 0], q=[0 0 0 1])
2025-03-20 12:25:14,395 - DEBUG - Created CartesianMotion: <franky._franky.CartesianMotion object at 0x7f8ac9119630> with ReferenceType: ReferenceType.Relative
2025-03-20 12:25:14,398 - DEBUG - Fehler beim Abrufen des aktuellen kartesischen Zustands: Invalid elbow configuration given! Only +1 or -1 are allowed for the sign of the 4th joint.
2025-03-20 12:25:14,426 - DEBUG - Current joint positions: [-2.43615962 -0.28623811  2.8617084  -1.67189819  0.30439842  1.74952786
  1.223974  ]
2025-03-20 12:25:14,429 - DEBUG - Robot IP: 192.168.0.103
2025-03-20 12:25:14,438 - DEBUG - Relative dynamics factor: <franky._franky.RelativeDynamicsFactor object at 0x7f8ac8a99570>
2025-03-20 12:25:14,440 - DEBUG - Target translation: [0.2, 0.0, 0.0]
2025-03-20 12:25:14,442 - DEBUG - Target affine: Affine(t=[0.2 0 0], q=[0 0 0 1])
2025-03-20 12:25:14,445 - DEBUG - Motion command: <franky._franky.CartesianMotion object at 0x7f8ac9119630>
2025-03-20 12:25:14,460 - DEBUG - ReferenceType: ReferenceType.Relative
2025-03-20 12:25:14,469 - DEBUG - Sending move command to robot
2025-03-20 12:25:14,510 - ERROR - Ein Fehler ist aufgetreten:
2025-03-20 12:25:14,527 - ERROR - Traceback (most recent call last):
  File "/home/johannes/Downloads/FrankyTest/test.py", line 57, in <module>
    robot.move(motion)
    ~~~~~~~~~~^^^^^^^^
franky._franky.RealtimeException: libfranka: unable to set realtime scheduling: Operation not permitted

(env) root@Fedora42:/home/johannes/Downloads/FrankyTest# /home/johannes/Downloads/FrankyTest/env/bin/python /home/johannes/Downloads/FrankyTest/test.py
2025-03-20 12:25:41,466 - DEBUG - Initializing robot with IP: 192.168.0.103
2025-03-20 12:25:41,475 - DEBUG - Set relative_dynamics_factor: <franky._franky.RelativeDynamicsFactor object at 0x7f53bf82ff30>
2025-03-20 12:25:41,475 - DEBUG - Target translation: [0.2, 0.0, 0.0]
2025-03-20 12:25:41,475 - DEBUG - Created target Affine: Affine(t=[0.2 0 0], q=[0 0 0 1])
2025-03-20 12:25:41,476 - DEBUG - Created CartesianMotion: <franky._franky.CartesianMotion object at 0x7f53bf898c30> with ReferenceType: ReferenceType.Relative
2025-03-20 12:25:41,477 - DEBUG - Error when retrieving the current Cartesian state: Invalid elbow configuration given! Only +1 or -1 are allowed for the sign of the 4th joint.
2025-03-20 12:25:41,480 - DEBUG - Current joint positions: [-2.4361597  -0.28624026  2.86170855 -1.67189283  0.3043983   1.74953107
  1.223974  ]
2025-03-20 12:25:41,480 - DEBUG - Robot IP: 192.168.0.103
2025-03-20 12:25:41,481 - DEBUG - Relative dynamics factor: <franky._franky.RelativeDynamicsFactor object at 0x7f53bf98d130>
2025-03-20 12:25:41,481 - DEBUG - Target translation: [0.2, 0.0, 0.0]
2025-03-20 12:25:41,481 - DEBUG - Target affine: Affine(t=[0.2 0 0], q=[0 0 0 1])
2025-03-20 12:25:41,481 - DEBUG - Motion command: <franky._franky.CartesianMotion object at 0x7f53bf898c30>
2025-03-20 12:25:41,481 - DEBUG - ReferenceType: ReferenceType.Relative
2025-03-20 12:25:41,482 - DEBUG - Sending move command to robot
2025-03-20 12:25:41,494 - ERROR - Ein Fehler ist aufgetreten:
2025-03-20 12:25:41,507 - ERROR - Traceback (most recent call last):
  File "/home/johannes/Downloads/FrankyTest/test.py", line 57, in <module>
    robot.move(motion)
    ~~~~~~~~~~^^^^^^^^
ValueError: Invalid elbow configuration given! Only +1 or -1 are allowed for the sign of the 4th joint.
```
Do the Current joint positions seem normal to you ? 

You suggested, that I could write print statements in the libfranka code.
The problem is right now that I dont know how to build custom wheels. Is there a way to build custom wheels in the Docker Container you provide ?

TimSchneider42, 2025-04-02:
Hey, 

sorry, I forgot to follow up on this. The instructions I wrote above build wheels for your custom code:

```bash
git clone --recurse-submodules https://github.com/TimSchneider42/franky/
cd franky
docker compose build franky-build --build-arg LIBFRANKA_VERSION=0.5.0
docker compose run --rm franky-build build-wheels
```

From what I can see, libfranka 0.5.0 handles elbows differently than the later version. Specifically, I think libfranka 0.5.0 does not allow 0 for the flip of the 4th joint, while later versions use that as a default to indicate that no elbow configuration was given. But I am not sure why that leads to the error you are experiencing.

Best,
Tim

EntrigesFichtenholz, 2025-06-21:
Hi, I haven't completely given up yet, but due to the exams I'm writing this semester, I haven't had much time. I'll pick up the topic again when circumstances allow. I've tried a lot to modify lib franka myself, but I've come to the conclusion that I'd rather write a translation layer when I have time again.

Thank you for your help,
Johannes

closed 2025-06-21 by EntrigesFichtenholz

Entongsu, 2025-08-13:
Hi, 

I have encountered a similar problem. Is there a version of Franky compatible with 0.5.0? Thank you.