TimSchneider42/franky #22: support joint velocity cotrol ?

wkinger, 2025-03-21:
I see that Franky supports joint position and cartesian position control. Does it support joint velocity control?

wkinger, 2025-03-21:
@TimSchneider42 Hello，Does Franky support joint velocity control mode?

wkinger, 2025-04-01:
@TimSchneider42 
This is  a delightful piece of news，and thanks a lot for your effort. But  unfortunately I met a problem when I try to install the newest version of franky by pip:
`pip install franky-panda==0.11.3 -i https://pypi.org/simple`
but versions higher than v0.9.1 could not be found by this way: 
```
ERROR: Could not find a version that satisfies the requirement franky-panda==0.11.3 (from versions: 0.1.0, 0.1.3, 0.2.0, 0.2.1, 0.3.0, 0.3.1, 0.3.2, 0.4.0, 0.4.1, 0.4.2, 0.4.3, 0.4.4, 0.5.0, 0.5.1, 0.6.2, 0.7.0, 0.7.1, 0.7.2, 0.8.0, 0.9.0, 0.9.1)
ERROR: No matching distribution found for franky-panda==0.11.3
```
I wonder if you know what is wrong, or provide a substitude way？Thanks a lot！

Best,
wkinger

TimSchneider42, 2025-04-01:
What is your python version?

wkinger, 2025-04-01:
python 3.12.7 in conda .Thanks for replying, I've just successfully  installed the newest version of franky from source. Now I am already to try out the joint/cartesian velocity control mode!

TimSchneider42, 2025-04-08:
I am in the process of publishing franky 1.0.0, which supports velocity control. It will be under a new package name `franky-control` on PyPi. If you want, you can give it a try and let me know if it works for you.

closed 2025-05-29 by TimSchneider42