TimSchneider42/franky #24: Franky-control segmentation faults with libfranka 0.13.3

lzyang2000, 2025-04-10:
Thank you for the great package! I've been using the older franky-panda 0.11.3 built for libfranka 0.13.3 fine, however when I upgraded it to franky-control 1.0.0, a segmentation fault would happen when I try to call robot.move(motion), i.e.

```python
from franky import Affine, CartesianMotion, Robot, ReferenceType

robot = Robot("172.16.0.2")
robot.relative_dynamics_factor = 0.05

motion = CartesianMotion(Affine([0.2, 0.0, 0.0]), ReferenceType.Relative)
robot.move(motion) #Segmentation fault (core dumped)
```
I'm wondering if there are any pointers to debugging this. 
Thank you again!

TimSchneider42, 2025-04-10:
Uh, that's not great. Could you test whether you can access the robot state without segfault? Just calling `print(robot.cartesian_pose)` should do. Also, does that happen with different motion types as well? E.g. `JointMotion`.

Best,
Tim

lzyang2000, 2025-04-11:
Thanks for the reply! I've performed some tests and the results are as follows:

tldr: 
most if not all of reading the state from the robot work, defining motions work, but it always segfaults when calling `robot.move(motion)`. I've put the error messages in the comments after the affected lines.
```python
from franky import *

robot = Robot("172.16.0.2")
robot.relative_dynamics_factor = 0.05

motion = CartesianMotion(Affine([0.2, 0.0, 0.0]), ReferenceType.Relative)
state = robot.state

cartesian_state = robot.current_cartesian_state
robot_pose = cartesian_state.pose  
ee_pose = robot_pose.end_effector_pose
elbow_pos = robot_pose.elbow_state # AttributeError: 'franky._franky.RobotPose' object has no attribute 'elbow_position', had to use elow_state
robot_velocity = cartesian_state.velocity  
ee_twist = robot_velocity.end_effector_twist
elbow_vel = robot_velocity.elbow_velocity
joint_state = robot.current_joint_state
joint_pos = joint_state.position
joint_vel = joint_state.velocity

# Use the robot model to compute kinematics
q = [-0.3, 0.1, 0.3, -1.4, 0.1, 1.8, 0.7]
f_t_ee = Affine()
ee_t_k = Affine()
# ee_pose_kin = robot.model.pose(Frame.EndEffector, q, f_t_ee, ee_t_k) #double free or corruption (out)

# Get the jacobian of the current robot state
# jacobian = robot.model.body_jacobian(Frame.EndEffector, state) #double free or corruption (out)

# urdf_model = robot.model_urdf # AttributeError: 'Robot' object has no attribute 'model_urdf'
m_jv1 = JointVelocityMotion([0.1, 0.3, -0.1, 0.0, 0.1, -0.2, 0.4], duration=Duration(1000))
m_jp1 = JointMotion(q)
m_jp2 = JointStopMotion()
m_cp1 = CartesianMotion(Affine([0.2, 0.0, 0.0]), ReferenceType.Relative)
m_cp2 = CartesianStopMotion()
# robot.move(m_jv1)
# robot.move(m_jp1)
# robot.move(m_jp2)
# robot.move(m_cp1)
# robot.move(m_cp2)
# all of the above produces Segmentation fault (core dumped)
```
Thank you!

TimSchneider42, 2025-04-11:
Great, thanks for testing. The issue seems to be related to the model class. Could you test if you run into any issues if you just call a model function? E.g.

```python
robot = Robot(...)

q = [-0.3, 0.1, 0.3, -1.4, 0.1, 1.8, 0.7] f_t_ee = Affine() ee_t_k = Affine() # ee_pose_kin = robot.model.pose(Frame.EndEffector, q, f_t_ee, ee_t_k)
```

Thanks a lot!

TimSchneider42, 2025-04-11:
I can reproduce the issue. Something is wrong with the model library. Let me fix it.

closed 2025-04-11 by TimSchneider42

TimSchneider42, 2025-04-11:
Okay, it turned out the model double-free error and the segmentation fault were two entirely different issues. I have fixed both now in version 1.0.1. It should be available in PyPi in around 20min.

lzyang2000, 2025-04-11:
can confirm problem solved, thanks!

commit 68506765 (2025-04-11): Fixed memory issue stemming from wrong holder type of Model binding. Fixes #24.
commit a49c6a11 (2025-04-11): Fixed waypoint iterator initialization in WaypointMotion. Fixes #24.