TimSchneider42/franky #25: Having manylinux_2_17 wheels

enyen, 2025-04-18:
Hi @TimSchneider42, it would be great to also have the wheels in manylinux_2_17 in the latest releases.
My system only runs on an older version of Ubuntu, which does not support manylinux_2_34. 
Bug (especially on Franka gripper in v0.9.1) fixes are the really important updates left behind in manylinux_2_17.
Thanks!

TimSchneider42, 2025-04-20:
Hey @enyen,

that is a great idea. I didn't really consider backward compatibility. Let me see what I can do after easter.

Best,
Tim

TimSchneider42, 2025-04-22:
Which Ubuntu version are you using? Do you need manylinux 2_17 or does 2_28 also do?

enyen, 2025-04-22:
My system works until 2_31.
Thanks!

TimSchneider42, 2025-04-22:
That's good, because 2.17 is not supported by GitHub anymore. I just fixed the compatibility issues with 2.28. Could you give it a go?

enyen, 2025-04-22:
```bash
docker compose run --rm franky-build run-tests
```
still having the same error 

> /code/ruckig
> does not contain a CMakeLists.txt file.

TimSchneider42, 2025-04-22:
Did you clone with --recurse-submodules?

enyen, 2025-04-22:
My bad...
Now it found ruckig but has build error:

> [ 96%] Building CXX ob
> ject CMakeFiles/_franky-lib.dir/python/bind_state_repr.cpp.o
> [ 98%] Linking CXX executable linear
> [ 98%] Built target linear
> [100%] Linking CXX shared module _franky.cpython-311-x86_64-linux-gnu.so
> lto-wrapper: warning: using serial compilation of 5 LTRANS jobs
> lto-wrapper: note: see the ‘-flto’ option documentation for more information
> [100%] Built target _franky-lib
> [ 22%] Built target ruckig
> [ 70%] Built target franky
> [ 98%] Built target _franky-lib
> [100%] Generating Python stubs...
> Traceback (most recent call last):
>   File "/code/custom_stubgen.py", line 68, in <module>
>     run(
>   File "/build/franky-stubs-gen/venv/lib/python3.11/site-packages/pybind11_stubgen/__init__.py", line 326, in run
>     QualifiedName.from_str(module_name), importlib.import_module(module_name)
>                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/opt/python/cp311-cp311/lib/python3.11/importlib/__init__.py", line 126, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 676, in _load_unlocked
>   File "<frozen importlib._bootstrap>", line 573, in module_from_spec
>   File "<frozen importlib._bootstrap_external>", line 1233, in create_module
>   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
> ImportError: libPocoNet.so.95: cannot open shared object file: No such file or directory
> make[3]: *** [CMakeFiles/franky-stubs.dir/build.make:71: CMakeFiles/franky-stubs] Error 1
> make[2]: *** [CMakeFiles/Makefile2:213: CMakeFiles/franky-stubs.dir/all] Error 2
> make[1]: *** [CMakeFiles/Makefile2:253: CMakeFiles/_franky.dir/rule] Error 2
> make: *** [Makefile:238: _franky] Error 2
>

TimSchneider42, 2025-04-23:
That was some strange path issue, which should be fixed now. Can you try again?

TimSchneider42, 2025-04-24:
I just uploaded some wheels for manylinux 2.28. You should be able to install it from PyPI now.

enyen, 2025-04-24:
It works. Thanks!

closed 2025-04-24 by enyen