TimSchneider42/franky #26: How to perform realtime incremental actions in cartesian space?

Wel2018, 2025-04-24:
Hello,
I want to control the horizontal or vertical movement of the end-effector by long-pressing the `WASD` keys on the keyboard. However, I found that `robot.move(motion, asynchronous = True)` cannot achieve incremental movement. Here is an simple test:

```python
from argparse import ArgumentParser
import time
import timeit
from franky import *

if __name__ == "__main__":
    parser = ArgumentParser()
    parser.add_argument("--host", default="172.16.0.2", help="FCI IP of the robot")
    args = parser.parse_args()

    # Connect to the robot
    robot = Robot(args.host)
    robot.recover_from_errors()

    # Reduce the acceleration and velocity dynamic
    robot.relative_dynamics_factor = 0.1

    # Go to initial position
    robot.move(JointMotion([0.0, 0.0, 0.0, -2.2, 0.0, 2.2, 0.7]))

    step = 0.05
    ASYNC = 1
    STEPS = 10

    while 1:
        direct = input("keyboard: ")
        if direct == "w":
            target = Affine([step, 0.0, 0.0]) # type: ignore
        elif direct == "s":
            target = Affine([-step, 0.0, 0.0]) # type: ignore
        elif direct == "a":
            target = Affine([0, step, 0.0]) # type: ignore
        elif direct == "d":
            target = Affine([0, -step, 0.0]) # type: ignore
        elif direct == "z":
            target = Affine([0, 0, step]) # type: ignore
        elif direct == "x":
            target = Affine([0, 0, -step]) # type: ignore
        else:
            continue
        print(target)
        
        motion_forward = CartesianMotion(target, reference_type=ReferenceType.Relative)

        t0 = timeit.default_timer()
        for _ in range(STEPS):
            print(f"move {target}...")
            robot.move(motion_forward, asynchronous=bool(ASYNC))
        t = round((timeit.default_timer()-t0)*1000, 2)
        print(f"timecost: {t} ms")
```

If the "asynchronous" is set to True, each execution will take approximately 200 - 400ms, which does not meet the requirements of realtime control. Do you have any good suggestions?

TimSchneider42, 2025-04-24:
Hi,

Commanding a new motion from Python might take a couple of milliseconds, but not 200 - 400ms.  Judging from your code, you are measuring the time it takes to command a loop of 10 move commands, for which 200 - 400ms seems realistic.

I am not quite sure I understand what you are doing here. Could you tell me why you're commanding each motion 10 times in a loop? You could instead just command a single motion with 10 times the step size and then wait for a couple of milliseconds to get the desired effect. 

Also, did you see that Franky now supports velocity control? Maybe you could also have a loop that commands CartesianVelocityMotions, which would mean that the robot always keeps moving until it receives the next command and does not stop at the next step.

Hope that helps!

Best,
Tim

Wel2018, 2025-04-27:
> Hi,
>
> Commanding a new motion from Python might take a couple of milliseconds, but not 200 - 400ms. Judging from your code, you are measuring the time it takes to command a loop of 10 move commands, [...]

Thank you for your detailed reply. I want to achieve the keyboard teleoperation effect in Isaac Sim, as shown in this video:

https://github.com/user-attachments/assets/a08808c1-818e-40bd-8322-782ff5033e77

Now I have implemented the keyboard teleoperation function using the asynchronous mode of CartesianVelocityMotion, which meets my requirements 😁.

Wel2018, 2025-04-27:
@TimSchneider42 Now I'm facing another problem. When using joint space motion, the execution time is relatively long (about 180ms). When I reduce the target step size (for example, changing 0.005 to 0.001), the execution time is approximately 80ms, but the movement becomes very slow. When I increase the relative_dynamics_factor, the speed increases, but there is a significant jerk in the movement. Here, my goal is to achieve real - time control of each joint angle at a relatively high frequency, similar to libfranka's writeOnce. Do you have any suggestions? I'm looking forward to your reply. Here is my code:

```python
parser = ArgumentParser()
parser.add_argument("--host", default="172.16.0.2", help="FCI IP of the robot")
args = parser.parse_args()

# Connect to the robot
robot = Robot(args.host)
robot.recover_from_errors()
robot.relative_dynamics_factor = 0.1

# Go to initial position
robot.move(JointMotion([0.0, 0.0, 0.0, -2.2, 0.0, 2.2, 0.7]))
time.sleep(1)

for i in range(1000):
    t0 = timeit.default_timer()
    m = JointMotion([0.005, 0.005, 0.0, 0.0, 0.0, 0.0, 0.0], ReferenceType.Relative)
    robot.move(m, asynchronous=bool(0))
    t = round((timeit.default_timer() - t0)*1000, 2)
    print(f"Time taken for move: {t:.2f} ms")

# robot.join_motion(5000)
print(f"{robot.state.q}")
robot.stop()
print("Robot closed.")
```

Output:
```
Time taken for move: 181.60 ms
Time taken for move: 179.84 ms
Time taken for move: 180.12 ms
Time taken for move: 180.02 ms
Time taken for move: 179.42 ms
Time taken for move: 180.73 ms
Time taken for move: 180.69 ms
Time taken for move: 180.72 ms
Time taken for move: 179.88 ms
...
```

TimSchneider42, 2025-04-27:
Hi, 

It takes so long because you are commanding a synchronous motion, which will only return after reaching the goal. To achieve what you want to achieve, I recommend using `CartesianVelocityMotion`:

(I have not tested this code, use with care)
```python
from argparse import ArgumentParser
import time
from pynput import keyboard
from franky import *

if __name__ == "__main__":
    parser = ArgumentParser()
    parser.add_argument("--host", default="172.16.0.2", help="FCI IP of the robot")
    args = parser.parse_args()

    # Connect to the robot
    robot = Robot(args.host)
    robot.recover_from_errors()

    # Reduce the acceleration and velocity dynamic
    robot.relative_dynamics_factor = 0.1

    # Go to initial position
    robot.move(JointMotion([0.0, 0.0, 0.0, -2.2, 0.0, 2.2, 0.7]))

    # Track which control keys are currently held
    pressed_keys = set()
    
    def on_press(key):
        """Add pressed WASD keys to the set."""
        try:
            if key.char in ('w', 'a', 's', 'd'):
                pressed_keys.add(key.char)
        except AttributeError:
            pass  # Ignore non-character keys
    
    def on_release(key):
        """Remove released WASD keys from the set; Esc stops the listener."""
        try:
            if key.char in ('w', 'a', 's', 'd'):
                pressed_keys.discard(key.char)
        except AttributeError:
            pass
        if key == keyboard.Key.esc:
            return False
    
    # Start keyboard listener in the background
    listener = keyboard.Listener(on_press=on_press, on_release=on_release)
    listener.start()

    vel = 0.05  # Linear velocity in [m/s]

    while True:
        # Build the action vector on each tick
        x = (-vel if 'a' in pressed_keys else 0) + (vel if 'd' in pressed_keys else 0)
        y = ( vel if 'w' in pressed_keys else 0) + (-vel if 's' in pressed_keys else 0)
        
        robot.move(CartesianVelocityMotion(Twist([x, y, 0.0])), asynchronous=True)

        time.sleep(0.05)          # 50 ms update interval
```

Let me know if that works.

Best,
Tim

closed 2025-04-28 by Wel2018