TimSchneider42/franky #28: JointStopMotion() is not Defined

sarthakdas, 2025-04-25:
Hi I am running: 

`from franky import *
import time
import math
from scipy.spatial.transform import Rotation

robot = Robot("172.16.0.2")  # Replace this with your robot's IP
robot.relative_dynamics_factor = 0.05
while True:
    m_jp1 = JointMotion([-0.3, 0.1, 0.3, -1.4, 0.1, 1.8, 0.7])
    print("YA YEEET")
    robot.move(m_jp1, asynchronous=True) #if not try async
    m_jp4 = JointStopMotion()`

and get this error: 

`m_jp4 = JointStopMotion()
NameError: name 'JointStopMotion' is not defined. Did you mean: 'JointMotion'?`

not sure how to solve this, do you support async joint position commands.

TimSchneider42, 2025-04-25:
Hi,

For me, this code works. Which version of Franky are you using? Could you try upgrading to the latest version of `franky-control`?

Asynchronous motions are supported via the `asynchronous` parameter of `Robot.move`. More information is in the README and the documentation.

Let me know if that works.

Best,
Tim

sarthakdas, 2025-04-28:
Hi, I am working with version 0.11.2, i can not pip install franky-control as it says no matching distribution found, ```
ERROR: Could not find a version that satisfies the requirement franky-control (from versions: none)
ERROR: No matching distribution found for franky-control
```

sarthakdas, 2025-04-29:
I am using: Ubuntu 20.04.6 LTS

TimSchneider42, 2025-04-29:
Version 1.0.2 provides wheels for Ubuntu 20. Can you try installing that from PyPi?

```bash
pip install franky-control==1.0.2
```

sarthakdas, 2025-04-30:
Running that command also results in ERROR: Could not find a version that satisfies the requirement franky-control (from versions: none)
ERROR: No matching distribution found for franky-control. I think something is wrong with the wheels?

TimSchneider42, 2025-04-30:
What is your processor architecture? Currently, I am providing wheels for x86_64 only.