TimSchneider42/franky #29: How to improve the speed of joint position control?

Wel2018, 2025-04-28:
Hello, 

When using joint space motion, the execution time is relatively long (about 180ms). When I reduce the target step size (for example, changing 0.005 to 0.001), the execution time is approximately 80ms, but the movement becomes very slow. When I increase the relative_dynamics_factor, the speed increases, but there is a significant jerk in the movement. 

> **My goal is to achieve realtime control of each joint angle at a relatively high frequency, similar to libfranka's writeOnce. I need to achieve the recording and reproduction of the trajectory. This requires a relatively high sampling frequency and control frequency, and also requires high positional accuracy of the joints.**

My input is probably like this:

```
8.799028964338746e-05,-0.7853572128158407,9.744812628042526e-05,-2.3562464993713754,3.497643154273149e-05,1.5708436714175793,0.7854168031981299,0.00476969825103879,0.00476969825103879
9.183042319513458e-05,-0.7853582974727955,9.74478015052005e-05,-2.3562447034217042,3.669174709477846e-05,1.5708470154045824,0.7854151798561354,0.00476969825103879,0.00476969825103879
8.741269500473755e-05,-0.7853460426768539,9.566511245680046e-05,-2.3562496067393504,3.3538277318167294e-05,1.5708456543932936,0.7854104458878516,0.00476969825103879,0.00476969825103879
9.012569545746032e-05,-0.7853606895370485,9.744777349101747e-05,-2.3562398387166317,3.150795223131166e-05,1.5708444266068406,0.7854213549806957,0.00476969825103879,0.00476969825103879
9.191225408908035e-05,-0.7853606895380004,0.00010873807731451922,-2.356239056908912,3.329358767551629e-05,1.5708445477782709,0.785412629623244,0.00476969825103879,0.00476969825103879
9.422097309336774e-05,-0.7853606378573497,9.952465684568684e-05,-2.356231894963313,3.308039248660798e-05,1.5708469826255331,0.7854172133152884,0.00476969825103879,0.00476969825103879
9.564901874405984e-05,-0.7853607071793627,0.00010202140888248167,-2.356225419165108,3.1480107538161674e-05,1.5708488130084177,0.7854183193823168,0.00476969825103879,0.00476969825103879
... 
(There are still 1000 lines left, and it has been recorded for 1s）
```

Do you have any suggestions? I'm looking forward to your reply. 

Here is my code:

```python
# Connect to the robot
robot = Robot(args.host)
robot.recover_from_errors()

robot.relative_dynamics_factor = 0.1

# Alternatively, you can define each constraint individually
# robot.relative_dynamics_factor = RelativeDynamicsFactor(velocity=0.1, acceleration=0.05, jerk=0.1)

# Or, for more finegrained access, set individual limits
# robot.translation_velocity_limit.set(3.0)
# robot.rotation_velocity_limit.set(2.5)
# robot.elbow_velocity_limit.set(2.62)
# robot.translation_acceleration_limit.set(9.0)
# robot.rotation_acceleration_limit.set(17.0)
# robot.elbow_acceleration_limit.set(10.0)
# robot.translation_jerk_limit.set(4500.0)
# robot.rotation_jerk_limit.set(8500.0)
# robot.elbow_jerk_limit.set(5000.0)
# robot.joint_velocity_limit.set([2.62, 2.62, 2.62, 2.62, 5.26, 4.18, 5.26])
# robot.joint_acceleration_limit.set([10.0, 10.0, 10.0, 10.0, 10.0, 10.0, 10.0])
# robot.joint_jerk_limit.set([5000.0, 5000.0, 5000.0, 5000.0, 5000.0, 5000.0, 5000.0])

# Go to initial position
robot.move(JointMotion([0.0, 0.0, 0.0, -2.2, 0.0, 2.2, 0.7]))
time.sleep(1)

###############################################################################
# !!! Simulate the real-time control of the joint space using the relative increment method
for i in range(1000):
    t0 = timeit.default_timer()
    m = JointMotion([0.005, 0.005, 0.0, 0.0, 0.0, 0.0, 0.0], ReferenceType.Relative)
    robot.move(m, asynchronous=bool(0))
    t = round((timeit.default_timer() - t0)*1000, 2)
    print(f"Time taken for move: {t:.2f} ms")

# robot.join_motion(5000)
print(f"{robot.state.q}")
robot.stop()
print("Robot closed.")
```

Output：
```
Time taken for move: 181.60 ms
Time taken for move: 179.84 ms
Time taken for move: 180.72 ms
Time taken for move: 179.88 ms
...
```

TimSchneider42, 2025-04-28:
This seems to be a duplicate of #26. Can I close it?

Wel2018, 2025-04-28:
> This seems to be a duplicate of [#26](https://github.com/TimSchneider42/franky/issues/26). Can I close it?

If you look closely, you'll find that this is not a duplicate question. This is another problem I encountered. But I've already solved it now. Thank you 😂.