TimSchneider42/franky #35: Calling `gripper.width` is very slow and blocking

alberthli, 2025-06-15:
Hi - I've found that querying the state of the gripper by calling `gripper.width` seems to always take about 0.1s for me (and sometimes inconsistently, two queries can happen in very close temporal proximity).

Example code:
```python
import time
from franky import Gripper

robot_ip = "172.16.0.2"
gripper = Gripper(robot_ip)
start_time = time.time()

while True:
    print(f"time: {time.time() - start_time:.3f} | gripper width: {gripper.width:.3f}")
```
Example output:
```
time: 0.000 | gripper width: 0.081
time: 0.082 | gripper width: 0.081
time: 0.182 | gripper width: 0.081
time: 0.194 | gripper width: 0.081
time: 0.282 | gripper width: 0.081
time: 0.382 | gripper width: 0.081
time: 0.394 | gripper width: 0.081
time: 0.482 | gripper width: 0.081
time: 0.582 | gripper width: 0.081
time: 0.594 | gripper width: 0.081
time: 0.682 | gripper width: 0.081
time: 0.782 | gripper width: 0.081
time: 0.794 | gripper width: 0.081
time: 0.882 | gripper width: 0.081
time: 0.982 | gripper width: 0.081
time: 0.994 | gripper width: 0.081
time: 1.082 | gripper width: 0.081
```

This blocks my data collection loop, which I would ideally like to run much faster.

TimSchneider42, 2025-06-15:
Hi,

Unfortunately, I cannot make `Gripper.width` faster, as it is limited by the time it takes the robot to respond.

However, I should release the GIL in this function as otherwise everything is blocked in those 100ms. I will do that in the next update.

In general, the Franka gripper is very limited and almost unusable for real-time control beyond 1 Hz in my experience. I recommend replacing it with a Robotiq or Robotis gripper.

Best,
Tim

alberthli, 2025-06-15:
Thanks, I figured as much. I'll swap out the default gripper with a different one - it's just useful for testing.

closed 2025-06-15 by alberthli