TimSchneider42/franky #37: Unexpected Oscillation During JointVelocityMotion Execution

Patrick15Yao, 2025-06-27:
Hi, thanks for the great library!

I'm encountering an issue when using `JointVelocityMotion`. While `JointMotion` and `CartesianMotion` work as expected, running velocity-based joint control causes the robot to oscillate and move back and forth unexpectedly.

This happens even with simple constant velocity input like:

robot.move(
    JointVelocityMotion(
        [0.05, 0, 0, 0, 0, 0, 0],
        duration=Duration(3000)
    )
)

I also confirmed this behavior by running `velocity_control.py` from the `examples` directory — the robot does not move smoothly in the commanded direction but instead oscillates. 

Additionally, I noticed that sometimes one joint may suddenly move faster than the commanded velocity, in an unexpected direction, which could lead to potentially unsafe behavior or even damage to the robot.

Is this expected behavior? Or is there an additional constraint or setup required to stabilize joint velocity control?(BTW, I have already updated to 1.1.0 version with libfranka as 0.9.2)

Thanks in advance!

TimSchneider42, 2025-06-27:
Another thing to look at might be the time step (`time_step`). If it is above 1ms for many steps, then that means that your CPU cannot keep up with the control cycle. This could certainly cause oscillation issues.

TimSchneider42, 2025-07-10:
Hi Patrick,

I see you have really dug into the code! A couple of notes from my side:

1. The assignment of velocity to `input_parameter.current_position` is intentional, as we are using ruckig to plan a trajectory for reaching a desired target _velocity_ and not _position_ here. Hence, velocity becomes position, acceleration becomes velocity, and jerk becomes acceleration. 
2. The `extrapolateMotion` function is a fallback that is supposed to imitate the Franka's internal behavior when a control frame is missed. In particular, when the robot does not receive a control signal in time, it assumes a constant acceleration model. The `extrapolateMotion` function tries to predict what the robot did in the last frame when it notices that a frame was missed (that is, the time step is greater than 1ms). Then we continue to control based on that information. It is possible that this function does not work properly or does not accurately reflect what the robot is doing in case of joint velocity control.
3. Do you see the same behavior when you reduce `robot.relative_dynamics_factor`?

Best,
Tim

TimSchneider42, 2025-07-10:
Hey Patrick,

I just reviewed some code, and I think there might be an issue in `VelocityWaypointMotion::extrapolateMotion`. In line 71, it should be 

```c++
auto vel = toEigenD<7>(input_parameter.current_position);
```

and not

```c++
auto vel = toEigenD<7>(input_parameter.current_velocity);
```

I guess I got confused by that whole ruckig position is Franka velocity thing myself. I just committed a patch to the dev branch. Could you test if it solves your issue?

Best,
Tim

Patrick15Yao, 2025-07-11:
Thanks I will try my best to look into it.