TimSchneider42/franky #39: Waypoint is not smoothed

ommmid, 2025-07-09:
Ran the simple example provided in tutorial for [Cartesian Position Control](https://github.com/TimSchneider42/franky#:~:text=%3D%20JointVelocityStopMotion()-,Cartesian%20Position%20Control,-import%20math%0Afrom) and noticed the robot stops at each waypoint. Is this an intended behavior? I would expect Ruckig to smooth the given waypoint for time optimal trajectory and then that gets feed into the controller but it looks like each waypoint is considered end of a trajectory with zero velocity. That leads me to believe that the waypoint is not smoothed or time parameteriez by Ruckig by default, is there a parameter to force that or how we correctly use `CartesianWaypointMotion` ?

TimSchneider42, 2025-07-10:
Hi,

This is intentional, as the free version of the Ruckig motion planner unfortunately does not support intermediate checkpoints the way you describe them. However, `CartesianWaypoint` allows passing a target velocity (twist) for waypoints. So you can do, e.g., spline interpolation, yourself, and pass each intermediate waypoint together with its desired velocity to Franky if you want the robot to move continuously. 

Best,
Tim

ommmid, 2025-07-10:
I see, thanks for the response. It would be great to mention in the tutorial that `CartesianWaypoint` currently is using the free version and won't be able to do the smoothing automatically for clarification. We can close this issue.