TimSchneider42/franky #40: Multi-Waypoint Motion about real time plan

TTZYTttzyt, 2025-07-16:
HI, here，I' wondering
If a trajectory has multiple waypoints and the commands are sent at a fixed frequency, will the performance ensure that each waypoint is reached, or if the second waypoint command is sent before reaching the first point, will the system restart movement toward the second point?"

TTZYTttzyt, 2025-07-17:
I'm glad to receive your reply. What I want to clarify is whether the joint control function is blocking—that is, does each target joint angle command need to be completed before sending the next one? I'm not sure if you understand what I mean. Apologies, but I may not be able to provide the code at the moment.

TimSchneider42, 2025-07-17:
Hi,

Every motion supports asynchronous execution. If you execute a motion with multiple waypoints asynchronously, it will reach each waypoint sequentially until you call `robot.move` again or a reaction is triggered. Then franky will immediately switch from the old motion to the new motion.

Does that answer your question?

Best,
Tim

TTZYTttzyt, 2025-07-20:
That's great, thank you for your reply! I'll give it a try.

closed 2025-07-20 by TTZYTttzyt