TimSchneider42/franky #41: Can JointPoseControl and CartesianPoseControl set time duration?

Patrick15Yao, 2025-07-17:
Dear Tim,
(1)May I ask is it possible for Franky to set time duration for JointPoseControl and CartesianPoseControl?
For example, move from pose A to pose B for 3000ms exactly?

(2)Btw, may I ask when we set robot.relative_dynamics_factor = 0.5, does this scale the limits (i.e. max velocity, acceleration, jerk), or does it directly set the actual velocity and acceleration values used in motion execution?

(3) when we simply call JointPoseControl and CartesianPoseControl, may I know who determines how fast they move to these poses? is it purely determined by Ruckig?

Thanks for your time answering my quesionts

Yours
Patrick

TimSchneider42, 2025-07-18:
Hey,

1. You can set a minimum time for individual waypoints:
```python
motion = franky.JointWaypointMotion(
    [
        franky.JointWaypoint([-0.3, 0.1, 0.3, -1.4, 0.1, 1.8, 0.7]),
        franky.JointWaypoint([0.0, 0.3, 0.3, -1.5, -0.2, 1.5, 0.8], minimum_time=franky.Duration(3000)),
        franky.JointWaypoint([0.1, 0.4, 0.3, -1.4, -0.3, 1.7, 0.9]),
    ]
)
```
The motion planner will always try to reach each waypoint as quickly as possible with the specified dynamic constraints. If the `minimum_time` parameter is set, it will take at least this amount of time. Hence, if your dynamic constraints allow, the robot should use exactly `minimum_time` and otherwise take longer.

2. Primarily, it will scale the limits. But since the motion planner tries to reach each waypoint as quickly as possible, it will definitely reach some sort of limit (jerk, acceleration, or velocity of any number of joints). Hence, by increasing or decreasing the relative dynamics factor, the execution speed of motions is always directly affected.

3. That is fully determined by ruckig. You have control over that by setting either `relative_dynamics_factor`, configuring the dynamic robot limits individually, or setting the `minimum_time` parameter.

I hope that answers your questions.

Best,
Tim

TimSchneider42, 2025-07-18:
That is right. It will always hit at least one constraint.

closed 2025-08-05 by TimSchneider42