TimSchneider42/franky #43: cartesian reflex from readme example not working. Version 1.1.1, libfranka 0.9.2

MartinPDU, 2025-07-28:
We tried the cartesian reflex from your ReadMe example and got the following error message:
franky._franky.ControlException: libfranka: Move command aborted: motion aborted by reflex! ["cartesian_reflex"]

What can we do to get the reflex working correctly?

BTW: The reflex in the example moves the arm in the wrong direction...

Panda Robot Version 4.2.2,
Franky wheel v1.1.1 with libfranka 0.9.2,
Ubuntu 20.04.6,
Python 3.8

kind regards,
Martin

MartinPDU, 2025-07-31:
Hi Tim,
you are right I mixed up the franka reflex and the franky reaction.

The reaction example is still not working with the above changes to the code:
```
robot.set_collision_behavior(100, 100)
motion_down = CartesianMotion(Affine([0.0, 0.0, 0.1]), ReferenceType.Relative) 
reaction_motion = CartesianMotion(Affine([0.0, 0.0, -0.1]), ReferenceType.Relative)
reaction = Reaction(Measure.FORCE_Z < -10.0, reaction_motion)
motion_down.add_reaction(reaction)
robot.move(motion_down) # move with collision
```

The reaction movement is not being executed and franka throws an ControlException.

kind regards,
Martin

manavkulshrestha, 2025-09-19:
Hi, has there been any update on this?

I'm trying to use this functionality to stop the robot when it gets some force from the table while moving down, but the cartesian_reflex triggers before the reaction gets a chance to. I ran the testing code you provided with some changes and I get this output:
```
Reaction fired
```
And these figures:
[image]
[image]

As best I can tell, the reaction callback is called, but the motion doesn't actually stop and run the reaction motion?

Here's the code:
```
from franky import *
import numpy as np
from matplotlib import pyplot as plt

robot = Robot('192.168.131.41')
gripper = Gripper('192.168.131.41')

HOME_Q = [-np.pi/2, -np.pi/4, 0., -3*np.pi/4, 0., np.pi/2, np.pi/4]
CAM_OFFT = np.array([0.065, 0., 0.044])
SPEED = 0.1
FORCE = 20

robot.relative_dynamics_factor = RelativeDynamicsFactor(
    velocity=0.2,
    acceleration=0.01,
    jerk=0.01
)

robot.recover_from_errors()
robot.move(JointWaypointMotion([JointWaypoint(robot.current_joint_state.position), JointWaypoint(HOME_Q)]))
gripper.open(SPEED)

# testing
traj1 = []
traj2 = []

def cb_motion1(robot_state: RobotState, time_step: float, rel_time: float, abs_time: float, control_signal: ControlSignalType):
    # This function will get called for every control update the robot does (1kHz) while the initial motion is active
    traj1.append((abs_time, robot_state.O_F_ext_hat_K))

def cb_motion2(robot_state: RobotState, time_step: float, rel_time: float, abs_time: float, control_signal: ControlSignalType):
    # This function will get called for every control update the robot does (1kHz) while the reaction motion is active
    traj2.append((abs_time, robot_state.O_F_ext_hat_K))

def cb_reaction(*args, **kwargs):
    # This function will get called once when the reaction fires
    print("Reaction fired")

robot.set_collision_behavior(50, 50)
motion_down = CartesianMotion(Affine([0.0, 0.0, 0.25]), ReferenceType.Relative) 
reaction_motion = CartesianMotion(Affine([0.0, 0.0, -0.2]), ReferenceType.Relative)
reaction = Reaction(Measure.FORCE_Z < -10.0, reaction_motion)
motion_down.add_reaction(reaction)

motion_down.register_callback(cb_motion1)
reaction_motion.register_callback(cb_motion2)
reaction.register_callback(cb_reaction)

try:
    robot.move(motion_down) # move with collision
except ControlException as e:
    pass

# convert to numpy for plotting
traj1_durations, traj1_forces = zip(*traj1)
traj1_durations = np.array([d.to_msec() for d in traj1_durations])
traj1_forces = np.array(traj1_forces)
traj2_durations, traj2_forces = zip(*traj2)
traj2_durations = np.array([d.to_msec() for d in traj2_durations])
traj2_forces = np.array(traj2_forces)

# save
np.savez('force_test_traj.npz',
    traj1_durations=traj1_durations,
    traj1_forces=traj1_forces,
    traj2_durations=traj2_durations,
    traj2_forces=traj2_forces,
)

# plot
plt.plot(traj1_durations, traj1_forces[:,2], label='motion')
plt.title('Z force during motion')
plt.show()

plt.plot(traj2_durations, traj2_forces[:,2], label='motion')
plt.title('Z force during reaction')
plt.show()
```