TimSchneider42/franky #49: Weird real-time control behavior in a simple sine-wave joint position control test

Ys-Jia, 2025-08-12:
Dear Tim,

I am especially interested in real-time joint position control of Franky. I performed a simple joint position control test at a frequency of less than 20Hz, focusing only on the 6th joint. However, multiple problems are emerging.

Below is my code snippet:
```
fr3 = FrankyController()
fr3.move_to_home()
home_positions = fr3.get_joint_positions()

joint_index = 6 # 0~6
amplitude = 0.2
ctrl_freq = 20
traj_len = 1200
joint_trajectory = amplitude * np.sin(np.linspace(0, 2 * np.pi, traj_len))
cmd, response, vel = [], [], []
for i in range(traj_len):
    start_time = time.perf_counter()
    # Add delta position to one joint
    new_positions = home_positions.copy()
    new_positions[joint_index] += joint_trajectory[i]
    
    fr3.set_joint_positions_velocities(new_positions)
    cmd.append(new_positions)

    elapsed = time.perf_counter() - start_time
    sleep_time = max(0, 1/ctrl_freq - elapsed)
    time.sleep(sleep_time)

    response.append(fr3.get_joint_positions())
    vel.append(fr3.get_joint_velocities())
    print(f"Elapsed time: {time.perf_counter() - start_time:.5f} seconds")

# set_joint_positions_velocities in the FrankyController class is simply defined as
def set_joint_positions_velocities(self, tgt_q, tgt_qd=None):
    """
    Move to target joint positions with optional target velocities.
    """
    if tgt_qd is not None:
        # Create joint state with velocity
        joint_state = franky.JointState(tgt_q, np.array(tgt_qd))
    else:
        joint_state = franky.JointState(tgt_q)
    
    motion = franky.JointMotion(joint_state)
    self.robot.move(motion, asynchronous=True)
```

Though I have set the `robot.relative_dynamics_factor = 0.05` globally, and I have computed the required velocity at each step is much smaller than the maximum velocity 0.05 * 2.61rad/s for the 6th joint; the discontinuity exception shows up frequently:
```
franky._franky.ControlException: libfranka: Move command aborted: motion aborted by reflex! ["joint_motion_generator_velocity_discontinuity", "joint_motion_generator_acceleration_discontinuity"]
```

Even though sometimes the exception luckily did not show up, I found a huge lag between the command joint position and the actual joint position shown below:
[image]

I don't know if my test pipeline for the real-time joint position control is incorrect, introducing the **Frequent Exceptions** and **Lag**, or if there are some limitations that I should keep in mind for Franky under real-time control mode. Would you provide an example of sine-wave joint position real-time control? Thanks a lot for your help in advance!

TimSchneider42, 2025-08-12:
Hey,

This looks like the robot is hitting its velocity limit. Your maximum commanded joint velocity is 0.2 (your amplitude), while the joint velocity limit is only 0.05 * 2.61 = 0.1305.  Could you try increasing the limit a little bit to see if that helps?  

Also, you do not set target velocities at the waypoints, right? So the motion planner will always try to reach the next target with 0 velocity.

Let me know if that helps.

Best,
Tim

TimSchneider42, 2025-08-13:
Sorry, just realized that the sin function is stretched and thus the velocity is much lower than what I calculated. Could you still try increasing the dynamic limits to see what happens? Also, why is the amplitude 1.0 in your plot and not 0.2?