TimSchneider42/franky #57: Best practice for setting motion limits (velocity, acceleration, jerk) in practice?

lif314, 2025-09-22:
Hi, thanks for the great work on `franky`!  

I ran into some issues when tuning the motion parameters for my Franka robot.  

For example, when setting limits like below:

```python
robot.relative_dynamics_factor = 0.05
````

sometimes I get reflex aborts such as:

```
franky._franky.ControlException: libfranka: Move command aborted: motion aborted by reflex! ["joint_velocity_violation"]
# or
franky._franky.ControlException: libfranka: Move command aborted: motion aborted by reflex! ["cartesian_reflex"]
```

I understand these reflexes are expected safety features from libfranka, but I’d like to know what **parameter settings work best in practice**.

* Should I only use `relative_dynamics_factor` instead of directly tuning velocity/acceleration/jerk?
* Are there recommended “safe” values for jerk/acceleration that avoid frequent reflex aborts?
* In your experiments, what configuration balances **safety** and **responsiveness** best?

Any practical guidance or example configs would be greatly appreciated 🙏

wengmister, 2025-09-22:
From my experiences, `cartesian_reflex` mostly arises from impact/joint torque violation - you will probably want to double check parameters under `setCollisionBehavior` and see if they can be relaxed (especially if you have a heavier end-effector attached)

I believe `relative_dynamics_factor` mostly affects trajectory parameterization - velocity, acceleration, jerk - if too aggressive you'll likely see `joint_velocity_discontinuity/joint_acceleration_discontinuity` instead of the velocity violation error. You will probably want to check elsewhere - but I may be wrong.

closed 2025-09-25 by lif314

Asunatan, 2026-09-07:
Hi @lif314,

I'm facing similar issues with unexpected reflex aborts and trying to fine-tune collision thresholds using `robot.set_collision_behavior`. In my setup, I have an end‑effector (an ultrasound probe) mounted on the last flange.

My questions:
1. The same code sometimes works fine, but other times it keeps throwing `joint_velocity_violation`. Why could that happen?
2. Would a force threshold of 10 N be a reasonable starting point for a robot with a typical lightweight end‑effector, or is it too conservative/risky?
3. How did you resolve this?

Thanks!